savoyhardwaregroup/speedometer #1: Wheel Rotation Sensor

NikolaiRuhe, 2019-02-24:
The speedometer backend uses a sensor circuit to measure wheel rotation. It does so by detecting periodic changes in the AC created by the hub dynamo.

NikolaiRuhe, 2019-02-25:
The first steps will be to learn more about the electric characteristics of the hub dynamo. There are a a couple of connected properties:

- AC voltage
- current
- frequency
- waveform

These are affected by:

- rotational speed
- electric load
- type of dynamo

We expect to see some kind of AC voltage whose period duration we can use to measure rotation speed.

NikolaiRuhe, 2019-02-25:
Von Ole:

Ein gängiger günstiger Nabendynamo ist z.B. dieser hier: Shimano DH-C3000
https://bike.shimano.com/de-DE/product/component/nexus-c3000-int7/DH-C3000-3N-QR.html

[image]

Technische Daten:
Spannung: 6 V
Leistung: 3 W
Einbaubreite: 100 mm (die Breite zwischen den Gabelbeinen; die Achse ist 108 mm lang)

So einen oder ein ähnliches Modell bringe ich nächste Woche mit, ebenso einen LED-Scheinwerfer.

In dem "Händlerhandbuch", das auf der Seite verlinkt ist (https://si.shimano.com/pdfs/dm/DM-GN0001-20-GER.pdf), habe ich noch folgendes gefunden (S. 139):

"Der Nabendynamo erzeugt eine sehr hohe Spannung. Berühren Sie niemals direkt die Anschlussklemme des Dynamos, während Sie fahren oder das Laufrad sich dreht. Das Berühren der Anschlussklemme des Dynamos kann einen elektrischen Schlag verursachen."

Ich weiß nicht, wie ernst man das nehmen muss. Heißt das, an den Anschlussklemmen entsteht eine höhere Spannung als 6 V? Keine Ahnung.

NikolaiRuhe, 2019-03-01:
Ich habe mal die Schaltung von heute in [Eagle](https://www.autodesk.com/products/eagle/overview) gezeichnet:

[image]

ole, 2019-03-01:
@NikolaiRuhe Super, danke!

NikolaiRuhe, 2019-03-04:
1. Langsame Drehung, keine Last am Dynamo:
- ca. ± 2.5V
- 2.4 Hz AC
- ca. 1.3 km/h (langsames Schieben)

[image]

NikolaiRuhe, 2019-03-04:
2. Mittlere Drehung, keine Last am Dynamo:
- ca. ± 30V
- 36 Hz AC
- ca. 20 km/h (mäßig schnelle Fahrt)

Auf Kanal 2 (hellblau) das Signal vom Reflektionssensor.

[image]

NikolaiRuhe, 2019-03-04:
3. Schnelle Drehung, keine Last am Dynamo:
- ca. ± 60V
- 55 Hz AC
- ca. 30 km/h (schnelle Fahrt)

[image]

NikolaiRuhe, 2019-03-05:
Die drei Bilder zeigen, dass das Signal nicht wie erwartet sinusförmig ist, sondern beim Nulldurchgang eine Abflachung aufweist. Zwei mögliche Erklärungen fallen mir ein:

- Ungleichmäßige Rotationsgeschwindigkeit beim manuellen Drehen durch die starken Rastpositionen.
- Die Geometrie der Spulen ist nicht kreisförmig, sondern rechteckig abgeflacht. Das könnte eine unregelmäßige Strominduktion bedeuten.

Jedenfalls ist weder die Ursache, noch die Wirkung für uns sonderlich relevant. Wir können das Signal trotzdem gut gleichrichten und messen.

NikolaiRuhe, 2019-03-06:
Die Messung des gleichgerichteten Signals hat mich anfangs verwirrt, da sie nicht wie erwartet dem nach oben geklappten Sinus entsprach. Sie sah eher wie ein um einen DC-Offset verschobener Sinus aus.

Nico hat erkannt, dass die Abweichung aus dem nicht sinusförmigen Eingangssignal resultiert. Zur Veranschaulichung habe ich obiges Bild aus dem Oszi in Photoshop bearbeitet:

[image]

Die negativen Halbwellen (rot) der Originalkurve (gelb) sind nach oben gespiegelt in hellblau darüber gelegt. Das resultierende Signal sieht wieder wie ein Sinus aus.

ole, 2019-03-07:
Wir haben heute noch mal systematische Messungen des Nabendynamos (a) bei verschiedenen Drehgeschwindigkeiten sowie (b) ohne und mit angeschlossener Last gemacht. Die Messergebnisse poste ich gleich. Im Endeffekt sehen die Bilder ähnlich aus wie die von @NikolaiRuhe oben, nur dass wir ein paar mehr Fälle abgedeckt haben.

ole, 2019-03-07:
1. Sehr schnelle Drehung (Akkuschrauber auf voller Leistung), keine Last am Dynamo:

- ca. ± 190V
- 230 Hz AC
- ca. 125 km/h (extrem schnelle Fahrt)

[image]

ole, 2019-03-07:
2. Zügige Drehung, keine Last am Dynamo:

- ca. ± 55V
- 55 Hz AC
- ca. 30 km/h (schnelle Fahrt)

[image]

ole, 2019-03-07:
3. Langsame Drehung, keine Last am Dynamo:

- ca. ± 6V
- 4 Hz AC
- ca. 2 km/h (langsames Schieben)

[image]

ole, 2019-03-07:
4. Sehr schnelle Drehung, mit Last am Dynamo (LED-Scheinwerfer)

- ca. ± 8V
- 230 Hz AC (wie 1.)
- ca. 125 km/h (extrem schnelle Fahrt) (wie 1.)

[image]

Das Signal ist ein abgerundetes Rechteck (senkrechte Flanken) und kein Sinus o.ä.

ole, 2019-03-07:
5. Zügige Drehung, mit Last am Dynamo (LED-Scheinwerfer):

- ca. ± 8V
- 55 Hz AC (wie 2.)
- ca. 30 km/h (schnelle Fahrt) (wie 2.)

[image]

ole, 2019-03-07:
6. Langsame Drehung, mit Last am Dynamo (LED-Scheinwerfer):

- ca. ± 55V
- 4 Hz AC (wie 3.)
- ca. 2 km/h (langsames Schieben) (wie 2.)

[image]

ole, 2019-03-07:
7. Bonus: Sehr schenelle Drehung, mit Last am Dynamo (**1 Ω Widerstand**):

- **ca. ± 1,3V**
- 230 Hz AC
- ca. 125 km/h (extrem schnelle Fahrt)
 
[image]