savoyhardwaregroup/speedometer #2: Test Rig

NikolaiRuhe, 2019-02-24:
Create a test rig with an actual hub dynamo. A motor to drive the dynamo and adjustable load would be nice.

ole, 2019-02-25:
Mein Fahrradhändler hat mir einen alten Nabendynamo geschenkt. Für den Testaufbau müssen wir uns in der Tat noch etwas überlegen. Man muss den Dynamo an den Achsenden ziemlich festklemmen/-schrauben, damit man ihn vernünftig drehen kann.

Es handelt sich um dieses Modell: [Shimano DH-3N32-NT](https://www.bike-discount.de/de/kaufen/shimano-dh-3n31-nt-nabendynamo-36-loch-731913)

Einbaumaß: 100 mm
Achse: Vollachse (kein Schnellspanner), Gewinde M9

Ein Patentrezept, wie wir das am besten machen, habe ich noch nicht.

Foto:

[image]

NikolaiRuhe, 2019-02-26:
Hier ist eine [Explosionszeichnung](https://si.shimano.com/pdfs/ev/EV-DH-3N31-NT-3545.pdf).

NikolaiRuhe, 2019-02-26:
Auf dem Bild sieht es so aus, als ob Dein Modell die 36 Loch (18 Speichen pro Flansch) Variante ist. Kannst Du das bestätigen, @ole?

ole, 2019-02-26:
Ja, 36 Speichen ist korrekt.

ole, 2019-02-28:
Hier ein Video von einem professionellen Test Rig auf einem Messestand: 
https://www.youtube.com/watch?v=frzVUkbe9BA

Die treiben den Dynamo über einen Keilriemen an, der Motor scheint unter dem Tisch zu sein. Die Verbindung vom Antriebsrad zum Dynamo scheint über die Scheibenbremsaufnahme des Dynamos zu laufen (wo normalerweise die Bremsscheibe montiert wird). Unser Dynamo hat keine Scheibenbremsaufnahme, aber @jjac meinte, man könnte ggf. die Speichenlöcher für etwas ähnliches verwenden.

[image]

NikolaiRuhe, 2019-02-28:
Ich habe mal gerechnet: Der Dynamo erzeugt nominal 3W, vielleicht 5W unter idealen Bedingungen. Wenn der Wirkungsgrad 50% ist (hoch geschätzt) muss die Achse mit mindestens 10W Motorleistung angetrieben werden. Wenn man noch Reibungsverluste und den Wirkungsgrad des Antriebs einrechnet, muss der Motor mindestens 20–30W Anschlussleistung haben.

So einen habe ich nicht hier. Daher habe ich erst mal eine Kurbel ausgedruckt, die direkt auf den Flansch geschraubt wird:

[image]

Müssen wir uns halt beim Kurbeln abwechseln :)

ole, 2019-02-28:
Wow, sieht super aus!

jjac, 2019-02-28:
Cool! Sehr schöne Minimalversion.

Ich habe noch zwei motor-getriebene Varianten im Kopf, die kann ich morgen erläutern.

jjac, 2019-03-01:
Ich habe eine meiner Überlegungen dazu mal skizziert:

[image]

Die Aufhängung für den Dynamo habe ich in der perspektivischen Ansicht weggelassen. Als Antriebsrad könnten wir wahrscheinlich [etwas von der Stange nehmen](https://www.otto.de/p/holzrad-mit-gummireifen-o-ca-85-mm-565765065/#variationId=752866195). Bei dieser skizzierten Variante ist zu berücksichtigen, dass eine nicht unerhebliche seitliche Kraft auf die Antriebswelle wirken wird, die der Antriebsmotor auch verkraften muss.

Eine alternative nicht skizzierte Überlegung geht dahin, die Kraft auf den Dynamo per Zahnrad zu übertragen. Dadurch entfiele die seitliche Kraft auf die Antriebswelle. Ein Zahnkranz wäre (ähnlich zu Nikolais Kurbel) auf den Dynamo geschraubt, auf die Motorwelle würde statt eines Rades ein Zahnrad gesetzt. So könnte man auch eine Getriebeunter- oder übersetzung besser zuschneiden.

Die Motordrehzahl sollte idealerweise (stufenlos) über einen Regler einstellbar sein. Ich weiß jedoch nicht, ob das einfach umzusetzen ist.

ole, 2019-03-01:
@jjac Sehr schicke Zeichnung! Finde ich super.

NikolaiRuhe, 2019-03-08:
Ich denke, die aktuelle Variante mit Akkuschrauber reicht fürs erste.

closed 2019-03-08 by NikolaiRuhe