savoyhardwaregroup/speedometer #3: Power Supply

NikolaiRuhe, 2019-02-24:
The speedometer backend will draw its power from the dynamo. We have to investigate hiw we can derive a stable 3.3 V DC supply from that. Do we need a small rechargable battery?

ole, 2019-04-10:
Siehe die Schaltungsentwürfe in #19.

NikolaiRuhe, 2019-04-29:
Wir haben beim letzten Termin festgelegt, dass eine ungepufferte Versorgung nicht ausreicht. Diese würde bei jedem Stopp die MCU zurücksetzen und damit verhindern, dass wir zusammenhängende Fahrten identifizieren können.

Daher haben wir die Verwendung eines Superkondensators zur Pufferung in Betracht gezogen.

NikolaiRuhe, 2019-04-29:
Hier ist die Ladekurve eines 1F Kondensators, den ich am Labornetzteil mit 20mA Begrenzung auflade.

Der gelbe Kanal ist die Spannung am Kondensator (0.5V/Teilung). Da wo der Pfeil mit der 1 hinzeigt ist 0V.

Der hellblaue Kanal ist der Strom (max 20mA).

Sehr gut sieht man die verzögerte Ladung nach erreichen der Zielspannung, bei der der Strom nur langsam abfällt.

[image]

NikolaiRuhe, 2019-04-29:
So sieht das Entladen an einem 100 Ohm Widerstand aus.

Bei diesem starken Entladestrom ist unser U(min) von 1.8V schon nach ca. 40 Sekunden erreicht.

Wir werden allerdings mit viel kleineren Strömen entladen. Bei voller BLE Last können wir wohl mit ca. 1mA rechnen, bei normaler BLE-Verfügbarkeit mit noch viel weniger.

[image]

NikolaiRuhe, 2019-05-06:
Jetzt sind endlich die Depletion-Mode MOSFETs von [LCSC](https://lcsc.com/) eingetroffen:

[image]

Ich habe einen [CPC3703](http://www.ixysic.com/home/pdfs.nsf/www/CPC3703.pdf/$file/CPC3703.pdf) auf ein breakout board gelötet ...

[image]

... und getestet:

[image]

Am Gate liegt ein Testsignal (magenta) aus dem Funktionsgenerator (-4V ... +1V, 1kHz Sägezahn).
Die Spannung aus dem Labornetzgerät an der Drain-Source-Strecke wird durch einen Widerstand begrenzt, an dem auch (gelb) der Strom gemessen wird.

Man sieht sehr schön, wie der MOSFET leitet, bis die Gate Spannung unter ca. -3V sinkt.

ole, 2019-05-11:
Meine Notizen vom Treffen am 10.05.2019 (Thema Spannungsregelung):

---

Gelb: NRF51822 beim BLE-Advertising (Zoom auf ein einzelnes Advertising-Paket).
Blau: Supercap, der die MCU mit Energie versorgt. 

[image]

---

Schaltung für die Spannungsregelung mit Supercap: 
[image]

Notizen:

- Normaler "klassischer" LDO: AMS1117. Kann z.B. I_Q = 3 mA Verluststrom haben. 
  OnSemi NCP170: Supereffizienter LDO mit IQ = 1 μA. Nachteil: schwer zu verarbeitem weil extrem klein.

- MCU könnte den MOSFET abschalten, wenn der Supercap voll ist. Und erst wieder anschalten, wenn eine gewissen Spannung unterschritten wird.

- Thermischer Schalter am Depletion-Mode MOSFET. Streng genommen vielleicht gar nicht nötig, wenn wir den Lade-Teil der Schaltung dynamisch über die MCU abschalten, wenn der Supercap voll ist.

- Supercap auf 5,5 V aufladen ist sinnvoll. Aber MCU verträgt nur 3,6 V. Mögliche Lösung: noch ein LDO (z.B. 2,8 V) hinter dem Supercap. Dieser LDO muss einen sehr niedrigen I_Q haben, weil er immer läuft, z.B. NCP170. Lassen wir aber erstmal weg.

- MCU kann Spannung am Supercap messen. Bei uns: Spannung = Ladezustand.

- MCU könnte auch die Eingangsspannung vor dem MOSFET messen (mit großem Spannungsteiler).

- Den Glättungskondensator vor dem MOSFET brauchen wir nicht mehr, weil wir den Supercap auch pulsartig laden können. Spart ein großes Bauteil.

- Vielleicht setzen wir einen Glättungskondensator vor den LDO, aber hinter den MOSFET (wo er keine 200 V aushalten muss).

- LDOs haben oft ein "Power good"-Signal.

- Wenn D1 direkt am Dynamo abgegriffen wird, messen wir am digitalen Eingang GPIO_S nur die positiven Halbwellen.

- Wenn wir D1 hinter dem Brückengleichrichter abgreifen, könnten wir auch die negativen Halbwellen als positive Signale messen. Aber damit verbauen wir uns die Möglichkeit, den Glättungskondensator doch noch einzusetzen.

- GPIO_D: Debugging-Ausgang mit angeschlossener LED. Von der MCU abschaltbar.
Debouncing können wir in Software machen oder mit einem schnellen Kondensator (RC-Glied).