savoyhardwaregroup/speedometer #8: Light Switch

NikolaiRuhe, 2019-02-25:
As the speedometer backend is expected to be mounted in the middle of the dynamo cable it would be relatively easy to include a light switch. The dynamo voltage would be only passed through the device when the (software) light switch is set to on.

The frontend could contain a physical switch or pish button to control the light. An ambient sensor could switch light on automatically.

This feature is completely orthogonal to the speedometer function and so has a relatively low priority.

ole, 2019-02-25:
Schöne Idee. Viele moderne Fahrrad-Scheinwerfer ab der mittleren Preisklasse haben selber einen Lightsensor, der das Licht automatisch einschaltet.

NikolaiRuhe, 2019-02-25:
Ah, gut. Das heißt, dass das mit nur einem Sensor schon ganz gut zu funktionieren scheint.

NikolaiRuhe, 2019-02-25:
Wir könnten am Display einen zusätzlichen Button „Lichthupe“ einbauen. Weiß nicht, ob das am Fahrrad sinnvoll ist?

ole, 2019-02-25:
Lichthupe finde ich eine lustige Idee. Müsste wahrscheinlich ein separater Schalter sein, den man mit der Hand am Lenker direkt dem Daumen bedienen kann (wie eine Klingel oder Schalthebel). Du denkst jedenfalls in die richtige Richtung. Es gibt inzwischen schon einige High-End-Fahrradscheinwerfer mit Fernlicht (Schalter am Lenker), was bei Waldfahrten etc. durchaus sinnvoll ist.

Meines Wissens sind diese Scheinwerfer aber bisher batteriebetrieben und hauptsächlich für E-Bikes gedacht, wo man Strom ohne Ende hat. Vielleicht sind die 3 W vom Nabendynamo für Fernlicht zu wenig.

NikolaiRuhe, 2019-02-25:
Hm, zusätzlicher Schalter, der per BLE Sonderfunktionen im Backend aktiviert? Elektrische Hupe?