savoyhardwaregroup/speedometer #12: Nächster Termin

NikolaiRuhe, 2019-03-02:
Ich bin am nächsten Freitag nicht in Berlin. Es gibt drei Optionen:

- Ausfallen lassen
- Verschieben auf Donnerstag
- Termin findet wie geplant statt (ohne mich)

Ich bin für Donnerstag – aber Ihr könnt auch gerne ohne mich am Freitag ins Labor :)

ole, 2019-03-02:
Mir passt auch Donnerstag. Freitag ist ja eh Feiertag in Berlin, wobei ich da trotzdem kommen könnte.

omz, 2019-03-03:
Mir würde Donnerstag auch passen.

strave, 2019-03-05:
Leider kann ich am Donnerstag nicht kommen. Mein Sohn ist seit gestern krank mit hohem Fieber. Ich vermute, dass er am Donnerstag noch nicht in die Schule gehen kann :(.

NikolaiRuhe, 2019-03-05:
> Leider kann ich am Donnerstag nicht kommen.

Schade! Ich nehme an, Freitag passt dann auch nicht?

Dann würde ich vorschlagen, dass wir Donnerstag einen nicht so langen Termin machen und kräftig protokollieren. Das wäre in jedem Fall hilfreich, denke ich.

NikolaiRuhe, 2019-03-05:
> Mein Sohn ist seit gestern krank

Gute Besserung, Noah!

strave, 2019-03-05:
@NikolaiRuhe, am Freitag sehen die Chancen besser aus, dass Noah wieder in die Schule gehen kann. Aber am Freitag bist Du nicht mitdabei. Ich denke ihr macht am Donnerstag weiter und ich lese dann fleissig die Berichte durch.

NikolaiRuhe, 2019-03-05:
Wenn Ihr Lust habt, würde ich Euch das Labor am Freitag überlassen und Ihr könntet nach Herzenslust experimentieren und messen.

Ich bin zwar auch sehr gerne mit dabei, aber Freitag kann ich ja nicht.

nsc, 2019-03-06:
Dann ist es beschlossen. Wir treffen uns morgen (Donnerstag) um 10:00 und versuchen nicht ganz so lang zu machen. Jörg und ich müssen nachmittags noch ein paar Zeilen Code für fremde Leute schreiben.

ole, 2019-03-06:
Okay, bis morgen!

NikolaiRuhe, 2019-03-13:
Ich habe folgende Vorschläge für den nächsten Termin:

- Auswerten der Messergebnisse
- Schaltung für Spannungsversorgung (5V, 50mA) planen und testen.
- Schaltung für Sensor (Frequenzmessung) verbessern und testen.
- Bonus: Arduino mit Signal versorgen und LED im Sensortakt blinken lassen.

Wie sieht's aus am 15.3.? Nico und Jörg können nicht. Sole, weißt Du schon, wie es bei Dir aussieht?

ole, 2019-03-13:
Ich hätte am Freitag Zeit. Den Teil mit dem Arduino könnte ich ggf. vorbereiten und zeigen — zumindest den Teil ohne die Verbindung mit der restlichen Schaltung, aber das ist dann auch nicht mehr so schwer. Ich habe so ein Arduino-Starter-Kit zu Hause.

NikolaiRuhe, 2019-03-13:
Ich habe eben mit Sole gesprochen: Sie kann Freitag nicht. Da ich auch nur eingeschränkt kann, schlage ich vor, dass wir diesen Freitag ausfallen lassen und dann nächsten Freitag wieder alle zusammen weiter machen.

NikolaiRuhe, 2019-03-13:
@ole Ja, das wäre super, wenn Du den Arduino mitbringst! Am besten so programmiert, so dass ein PIN (z.B. GPIO 2)  als Input auf einen anderen PIN als Output (Z.B. GPIO 13 == LED) kopiert wird.

ole, 2019-03-13:
Okay, das mache ich. Dann bis zum 22.03.

NikolaiRuhe, 2019-03-18:
Wie sieht‘s bei Euch am Freitag (22.3.) aus?

jjac, 2019-03-18:
Ich hab‘s eingeplant. 

Viele Grüße, Jörg.

ole, 2019-03-18:
Ich auch. Arduino ist auch vorbereitet, bringe ich mit.

strave, 2019-03-18:
Ich habe immer noch einen sehr hartnäckigen Husten aber ich hoffe ich bin bis Freitag wieder gesund.

nsc, 2019-03-20:
Ich bin auch dabei am Freitag. Gute Besserung, Sole!

omz, 2019-03-20:
Ich bin grad bei meinen Eltern auf dem Land und wahrscheinlich auch nächste Woche nicht in Berlin, kann also leider nicht dabei sein, bin aber gespannt auf eure Fotos usw.

NikolaiRuhe, 2019-03-20:
Ok, dann schlage ich vor, wir treffen uns am Freitag um 10, wie gehabt.

ole, 2019-03-20:
Und ich bringe dieses Mal das Frühstück mit, okay? @NikolaiRuhe @nsc Geht ihr immer zu einem bestimmten Bäcker in der Nähe?

NikolaiRuhe, 2019-03-20:
Super! Ich hole die Sachen immer bei dem Italiener Sarrazinstr.-Ecke-Bundesallee.

NikolaiRuhe, 2019-03-26:
Wie sieht's bei Euch am Freitag (29.3.) aus?

NikolaiRuhe, 2019-03-26:
Themen:

- Recap Spannungsversorgung
- Planung Display
- Kickoff MCU
    - Hardwarearchitektur des SoC
    - Toolchain und Anschluss an den Mac
    - Softwaremodule definieren (und verteilen?)

Als Ergebnis sollte jeder zu Hause auf eigener Hardware entwickeln und testen können.

jjac, 2019-03-26:
Ich bin dabei. 

Viele Grüße, Jörg.

ole, 2019-03-26:
Ich muss es offen halten. Ich versuche zu kommen, kann es aber erst am Donnerstag sicher sagen.

strave, 2019-03-27:
Ich bin am diesen Freitag dabei. Es sei den die Kinder bringen wieder ein "Geschenk" aus der Kita/Schule mit. Die Erzieherin hat mich davor gewarnt, dass es jetzt Magen-Darm und Läuse gibt 😩...

NikolaiRuhe, 2019-03-28:
Der morgige Termin findet statt. Es wäre gut, wenn Ihr Eure Mac Books mitbringt.

omz, 2019-03-28:
Ich bin noch nicht wieder in Berlin.

ole, 2019-03-28:
Ich werde es leider nicht schaffen. Euch viel Spaß!

strave, 2019-03-29:
Der Husten ist leider immer noch nich weg 😟. Ich bleibe lieber zuhause und inhaliere fleißig. Euch viel Spaß!!

NikolaiRuhe, 2019-04-01:
Ich bin am Freitag (voraussichtlich) bis 12.30 Uhr beschäftigt. Könnt Ihr auch nachmittags?

ole, 2019-04-01:
Ich kann auch am Nachmittag.

nsc, 2019-04-02:
Wir könnten uns norma um 10:00 Uhr treffen und dann eine Art Wiederholungstermin machen. Wir haben auch ein paar NRF51822-Boards und Programmer da und könnten gemeinsam die Rechner aufsetzen und ein erstes Testprogramm deployen.

ole, 2019-04-02:
Das finde ich eine sehr gute Idee.

jjac, 2019-04-02:
Sehr gut. Ich bin dabei.

omz, 2019-04-03:
Das finde ich auch eine sehr gute Idee und bin dabei, habe ja auch etwas Nachholbedarf...

strave, 2019-04-04:
Ich muss morgen schon um 11:15 Uhr los. Ist es Euch zu früh, wenn wir um 9:30 Uhr anfangen?

nsc, 2019-04-04:
Von mir aus könnten wir auch um 9:30 anfangen.

P.S.
Vergesst Eure Rechner nicht, wenn ihr die NRF-Toolchain installieren und ausprobieren wollt :)

jjac, 2019-04-04:
Ich kann auch 9:30.

NikolaiRuhe, 2019-04-10:
Ich bin Freitag wieder dabei. Wie sieht's bei Euch aus? 10 Uhr?

Habt Ihr Themenvorschläge/-wünsche?

jjac, 2019-04-10:
Ich bin dabei.

Themenvorschläge:
- Sensordaten mit Taster simulieren
- mbed:DigtalIn.read() Source-Code finden / verstehen
- Projekt-Repo anlegen (inkl. Xcode-Anbindung und eventuell VS Code-Anbindung)

Viele Grüße, Jörg

> Am 10.04.2019 um 15:46 schrieb Nikolai Ruhe <notifications@github.com>:
> 
> Ich bin Freitag wieder dabei. Wie sieht's bei Euch aus? 10 Uhr?
> 
> Habt Ihr Themenvorschläge/-wünsche?
> 
>

NikolaiRuhe, 2019-04-10:
VS Code Integration würde mich auch interessieren.

ole, 2019-04-10:
Ich bin auch dabei. Ich finde, wir sollten auf jeden Fall ein gemeinsames Projekt hier im Repo anlegen, an dem dann alle arbeiten können. Ich kann bei Bedarf noch was zu meiner kleinen Timerschaltung erzählen, um die Dynamosignale zu emulieren (wobei man das auch bei Ben Eater nachgucken kann).

Mit VS Code haben wir letzte Woche ja schon rumexperimentiert. Ich habe noch ein bisschen weitergemacht, war aber nicht erfolgreich. Bei mbed kann man das Projekt ja auch im CMake-Format runterladen. Das wäre cool, weil VS Code aus der CMake-Config direkt die Compiler-Konfiguration (für Code Completion etc.) auslesen kann. Leider funktioniert bei mir der Build nicht (d.h. CMake selbst funktioniert, aber der Build mit dem Makefile, das CMake generiert, schlägt fehl).

strave, 2019-04-11:
Mein Sohn ist leider wieder krank. I feel like I'm living in a continuous loop of illnesses... :(

NikolaiRuhe, 2019-04-11:
Gute Besserung!

ole, 2019-04-11:
Ja, gute Besserung!

NikolaiRuhe, 2019-04-11:
Falls Ihr Interesse habt, könnten wir auch mal ein bisschen in Richtung Elektronik-Prototyping gehen:

[image]

Ein kleiner NE555-basierter [Rechteckgenerator](http://www.ne555.at/2014/index.php/timer-ic-ne555/grundschaltungen/146-astabile-kippstufe-multivibrator-rechteckgenerator) als Dynamosimulator für die Firmwareentwicklung lässt sich schnell bauen. Den könnt Ihr an Euer nRF51822-Board anschließen und damit langsame (10Hz) bis schnelle (100Hz) Fahrt simulieren.

Topics:

- Mini-Crash-Kurs Schaltplanentwurf in Eagle
- Mini-Crash-Kurs in PCB-Design in Eagle
- Aufbau der Platine (endlich Löten!)
- Testen am Oszi

Ich habe alle Teile da (und zwei Lötkolben). Dann habt Ihr was zum mitnehmen und wir haben endlich mal was gebastelt.

NikolaiRuhe, 2019-04-24:
Wie sieht‘s aus am Freitag?

ole, 2019-04-24:
Ich bin dabei.

jjac, 2019-04-24:
Bin auch dabei.

Viele Grüße, Jörg.

nsc, 2019-04-25:
Ich bin auch dabei.

omz, 2019-04-25:
Bin auch dabei.