savoyhardwaregroup/speedometer #14: Treffen 8.3.2019

NikolaiRuhe, 2019-03-08:
Wir haben folgende Punkte besprochen:

- __Aufbau des Testrig:__ Die manuelle Version mit Kurbel drehte sich zu ungleichmäßig und langsam. Jetzt benutzen wir einen Akkuschrauber als Antrieb. Der erste Adapter ist nach kurzer Zeit gebrochen, da die Achse nicht flexibel genug war. Jetzt haben wir eine ein Meter lange Achse, die flexibel genug ist, um Ungenauigkeiten aufzunehmen.
- __Stromversorgung:__ Wir haben in Voraussicht auf die Stromversorgung der MCU über das Verhalten des Dynamos bei unterschiedlichen Lasten gesprochen. Es gibt hier immer wieder Unsicherheiten bei dem Thema Spannungs- vs. Stromversorgung. Ich habe noch mal vorgeschlagen, etwas Zeit für die grundlegenden Bauelemente und Gesetze zu investieren (#10).
- __MCU:__ Da das Schaffen einer Programmierumgebung für die Nordic MCU doch ziemlich aufwendig und komplex ist, habe ich dafür gestimmt, erst mal mit einer einfachen MCU (z.B. Arduino mit Atmel) zu beginnen. Wir können da ein einfaches Display anschließen und haben viel schneller ein Ergebnis. Der kabelgebundene Tacho wäre dann auch schon fertig. Die BLE Variante können wir als v2.0 sehr gut daraus weiterentwickeln.

Nach der Diskussion haben wir noch mal die Dynamospannung gemessen und dokumentiert. Wir haben mit und ohne Last (Lampe) und bei unterschiedlichen Geschwindigkeiten gemessen. Damit haben wir hoffentlich den gesamten Bereich abgedeckt, den wir bei der Dimensionierung der Schaltung berücksichtigen müssen:

- Amplitude der Spannung: ±4V – ±190V

Die Bauteile müssen also mindestens 200V Eingangsfestigkeit haben – aber auch bei 4V noch funktionieren.

NikolaiRuhe, 2019-03-13:
Damit es beim nächsten Termin etwas schneller vorankommen, habe ich noch ein paar Messungen mit Last gemacht. Hier sind die Ergebnisse:

[image]

[Numbers Dokument](https://github.com/savoyhardwaregroup/speedometer/files/2961438/Messung.numbers.zip)

jjac, 2019-03-13:
Wow, sehr gelungen.

ole, 2019-03-13:
Wirklich toll, danke!

NikolaiRuhe, 2019-03-13:
Hm, ohne Achsenbeschriftung nicht sehr hilfreich:

- x: f/Hz
- y: U(RMS)/V