savoyhardwaregroup/speedometer #17: Was ist der Unterschied zwischen Laborkarte und Platine?

ole, 2019-03-21:
Bei Reichelt gibt es so genannte "Laborkarten", z.B. [Laborkarte FR4 RM 1,27 mm 38,10 x 39,37 mm](
https://www.reichelt.de/laborkarte-fr4-rm-1-27-mm-38-10-x-39-37-mm-re-012lf-p167685.html).

Kann mir jemand erklären, was das ist? Sieht ja ähnlich aus wie eine [Lochraster-Platine](https://www.reichelt.de/lochrasterplatine-hartpapier-100x100mm-hpr-100x100-p8636.html), aber das ist eine separate Kategorie.

NikolaiRuhe, 2019-03-21:
Meistens heißt "Laborkarte", dass die Platine Euro-Maße hat (160 x 100 mm). In diesem Fall aber wohl nicht. Sonst kenne ich keine von "Platine" oder "Leiterplatte" abweichende Bedeutung.

FR4 steht für "Flame Retardant 4" und ist bei Platinen die Standardbezeichnung für das Trägermaterial aus Glasfaser + Epoxidharz. Es gibt noch FR2 (Hartpapier) und verschiedene Spezialträger (Alu, Flexibel, ...).

Das Rastermaß "RM 1,27" (= 1/20 Inch) ist unüblich (und doppelt so fein wie normal).

ole, 2019-03-21:
@NikolaiRuhe Danke.