savoyhardwaregroup/speedometer #18: Sourcing/Einkaufen

ole, 2019-03-21:
Fragen:

* Welche Teile brauchen wir?
* Wo kann man sie kaufen?
* Was gehört zu einer anständigen Grundausstattung (Teilesortiment etc.)?

ole, 2019-03-21:
Ich habe über ebay einen Shop namens Eckstein gefunden, der den Nordic 51822 und auch das Waveshare-Entwicklerboard führt uns aus Deutschland verschickt. Die verkaufen nicht nur über ebay, sondern haben auch einen eigenen Online-Shop.

* [NRF51822](
https://eckstein-shop.de/Core51822)
* [Waveshare NRF51822 Eval Kit](
https://eckstein-shop.de/NRF51822-Eval-Kit)

Ich habe diese Teile und noch einiges andere dort gestern mal bestellt. Bis morgen kommt das nicht mehr an, aber ich werde berichten.

NikolaiRuhe, 2019-03-21:
Ich habe mir die BLE400 Karte noch mal genauer angesehen. Die hat den eingebauten USB-UART leider so angeschlossen, dass nicht direkt damit programmiert werden kann (falscher Port).

Ihr braucht also noch einen Programmierer. Hier gibt es zwei Optionen:

- Ein universeller USB-Adapter (serielle Schnittstelle).
- Ein spezieller ARM Programmierer (ST-Link V2).

Für den USB-Seriell Adapter spricht, dass man ihn sowieso gut gebrauchen kann, da damit viele Boards programmiert werden (z.B. den von mir präferierten [Arduino Pro Mini](https://store.arduino.cc/arduino-pro-mini)).

Der ST-Link unterstützt dafür neben dem Programmieren auch das Debuggen, was extrem praktisch ist. Außerdem können damit viele ARM MCUs programmiert und debugged werden, nicht nur der Nordic.

Ich empfehle, dass wir in unserem Projekt mit dem ST-Link V2 arbeiten. Da beide Adapter sehr billig sind, spricht auch wenig dagegen, einfach beide zu kaufen.

NikolaiRuhe, 2019-03-21:
Zum Einkaufen können wir morgen ja noch mal sprechen. Mich interessiert zum Beispiel das Thema China-Bestellungen und Eure Meinung/Einwände.

ole, 2019-03-21:
> Mich interessiert zum Beispiel das Thema China-Bestellungen und Eure Meinung/Einwände.

Für mich persönlich ist es zum einen eine Frage der Bequemlichkeit:

* Wenn ich etwas bestelle, will ich nicht 3 Wochen darauf warten.
* Ich will Pakete meist an eine Packstation geliefert bekommen und nicht nach Hause oder ins Büro, vor allem wenn ich den Liefertag nicht voraussagen kann. Bei Lieferungen aus dem Ausland ist das meistens keine Option.
* Ich will das Risiko nicht eingehen, dass ich nach Schöneberg zum Zoll muss.

Zum anderen gibt es auch im weitesten Sinne ethische/moralische Aspekte:

* Anscheinend ist Umsatzsteuer-Hinterziehung bei vielen Nicht-EU-Händlern gang und gäbe, selbst wenn sie auf etablierten Plattformen wie Amazon verkaufen. Ich kann mir kaum vorstellen, dass das bei Aliexpress anders ist. Das will ich nicht unterstützen.
* Mein Gefühl ist, dass es für meinen persönlichen CO₂-Abdruck nicht gut sein kann, mit alle paar Wochen ein Paket aus China schicken zu lassen. Beziffern kann ich das aber nicht. Mir ist schon klar, dass die Sachen, die ich bei einem deutschen Händler kaufe, letztendlich auch mit demselben Containerschiff kommen.