savoyhardwaregroup/speedometer #19: Treffen 22.3.2019

NikolaiRuhe, 2019-03-22:
Heute haben wir uns drei Punkte vorgenommen:

1. Die Spannungsversorgung
2. Die Schaltung für den Drehfrequenzsensor
3. Inbetriebnahme der MCU

Mit der Spannungsversorgung haben wir uns am längsten beschäftigt. Es galt das Problem des großen Bereichs der Eingangsspannung zu lõsen und dabei nicht zu viel Verlustleistung (Abwärme) zu produzieren.

Die Messdaten vom letzten Treffen (#14) zeigen, dass wir mit Wechselspannungen von 4–200V klarkommen müssen. Erzeugen wollen wir eine geregelte Gleichspannung von 3,3V bei max. 20mA Last. Wir haben verschiedene Varianten dafür diskutiert (Zenerdiode, Shuntregler, Längsregler) und sind nun bei einer Lösung, die einen linearen Spannungregler benutzt, dessen Eingangsspannungsbereich durch einen MOSFET Transistor auf 500V erweitert wird.

... to be continued

NikolaiRuhe, 2019-03-23:
### Spannungsversorgung

Mit der Spannungsversorgung haben wir uns am längsten beschäftigt. Es galt das Problem des großen Bereichs der Eingangsspannung zu lõsen und dabei nicht zu viel Verlustleistung (Abwärme) zu produzieren.

Die Messdaten vom letzten Treffen (#14) zeigen, dass wir mit Wechselspannungen von 4–200V klarkommen müssen. Als Ziel wollen wir eine geregelte Gleichspannung von 3,3V bei max. 20mA Last erzeugen (Anforderung von der MCU).

Dies ist der aktuelle Stand der Spannungsversorgung:

[image]

Erklärung:

1. Der Input vom Dynamo ist Wechselspannung, ca. 4–200 V(RMS):

[image]

2. Der Brückengleichrichter aus D1-D4 macht daraus eine pulsierende Gleichspannung:

[image]

3. Um die nachfolgenden Bauteile zu schützen, begrenzen wir die Spannung durch die zwei Zenerdioden auf ca. 11 V:

[image]

4. Die immer noch pulsierende Spannung wird durch C1 geglättet. Eine Restwelligkeit verbleibt:

[image]

5. Der lineare Spannungsregler AMS1117 macht die resultierende Eingangsgleichspannung (ca. 4V–11V) zu geregelten 3.3V, mit denen wir dann die MCU versorgen können:

[image]

NikolaiRuhe, 2019-03-23:
Mit obigem Schaltung gibt es noch das Problem, dass der Dynamo bei hohen Geschwindigkeiten (und ausgeschaltetem Licht) sehr viel höhere Spannungen als 11V erzeugt. Der gesamte Rest wird durch die Zenerdiode abgeleitet und verursacht eine hohe Verlustleistung. Das erzeugt einerseits einen hohen Rollwiderstand und außerdem so viel Wärme, dass die Zenerdioden nach kurzer Zeit durchbrennen würden.

Um das Problem zu lösen, haben wir eine Schaltung aus "Art of Electronics" besprochen. Dabei wird die Spannung nicht durch Kurzschluss begrenzt, sondern ein weiterer Längsrichter einfügt, der die hohen Spannung aushält.

Es fließt also nicht mehr der Kurzschlussstrom durch die Zenerdioden, sondern nur noch der sehr viel kleinere Laststrom (max. 20mA). Die entstehende Verlustleistung liegt dann auch bei hohen Geschwindigkeiten noch in einem akzeptablen Bereich.

ole, 2019-03-24:
@NikolaiRuhe Danke für die tolle Zusammenfassung und Erklärung!

Ich ergänze noch ein paar Notizen, die ich mir gemacht habe und die vielleicht von Interesse sind (ob das alles 100% richtig ist, weiß ich nicht):

* Für die MCU (plus weitere Komponenten wie z.B. eine LED) planen wir folgenden Energieverbrauch ein:
  * Spannung: _V_CC = 3,3 V_
  * maximaler Stromverbrauch: _I_L = 20 mA_
  * ⇒ _P = 3,3 V * 0,02 A = 0,066 W < 0,1 W_ = sehr wenig

* Der lineare Spannungsregler ist ein [Low-dropout regulator (LDO)](https://en.wikipedia.org/wiki/Low-dropout_regulator). Er hat folgende Eigenschaften:
  * Regelt die Spannung runter auf 3,3 V und lässt dabei fast den vollen Strom durch.
  * Der LDO läuft immer, auch wenn die MCU schläft. Effizienz ist also sehr wichtig.
  * Am LDO abfallende Spannung: _V_LDO = V_In − V_Out_
    _V_Out = 3,3 V_
    _V_In_ kann ohne vorgeschaltete Begrenzung bis zu 200 V sein (vom Dynamo).
  * LDO produziert Verlustleistung (= Wärme): _P_Verlust = V_LDO * 20 mA_
    Extremfall: _P_Verlust = ~200 V * 20 mA = 4 W_
    Realistischer (bei 25–30 km/h): _P_Verlust = ~50 V * 20 mA = 1 W_. Das ist akzeptabel.
  * Es gibt allerdings sowieso keinen LDO, der so eine große Regelungsspanne (200 V auf 3,3 V) hat. Wir brauchen also sowieso eine vorgeschaltete Spannungsbegrenzung (Zenerdiode oder besser MOSFET, siehe oben).

* Schaltung mit [MOSFET (metal-oxide-semiconductor field-effect transistor)](https://en.wikipedia.org/wiki/MOSFET):
  * MOSFET kann **leistungslos** schalten (es fließt kein Strom zwischen Gate und Source).
  * [Depletion-mode MOSFET](https://en.wikipedia.org/wiki/MOSFET#Depletion-mode) = der Transistor ist bei 0 V leitend und wird sperrend, wenn die Spannung zwischen Gate und Source (V_GS) einen bestimmten (negativen) Wert unterschreitet.
  * In unserer Schaltung: sobald die Eingangsspannung am LDO 4 V überschreitet, fängt der LDO an zu regeln. Dadurch wird _V_GS_ negativ: am Gate liegen 3,3 V an, an der Source 4 V oder mehr. Ab −2,5 V Spannungsdifferenz sperrt der MOSFET und begrenzt dadurch die Eingangsspannung am LDO (Feedback-Loop).

* Wir können zur Sicherheit trotzdem noch einen thermischen Schutzschalter einbauen, der die gesamte Schaltung direkt hinter dem Nabendynamo unterbricht, wenn der MOSFET zu warm wird.

[image]

ole, 2019-03-24:
### Schaltung für den Drehfrequenzsensor

Ziel: das Wechselstrom-Signal vom Nabendynamo mit bis zu 200 V in ein digitales Signal von 0–3,3 V umwandeln, das wir an einen GPIO-Pin der MCU anschließen können.

Die Schaltung für den Drehfrequenzsensor ist in diesem Bild der grün markierte Teil:

[image]

Die zwei Dioden auf GND und _V_DD_ clampen die Spannung von max. ±200 V auf 0–3,3 V. Vorgeschaltet ist ein sehr großer 1 MΩ Widerstand, um den Strom auf ein Minimum zu begrenzen.

NikolaiRuhe, 2019-03-24:
@ole Prima, danke für die Ergänzungen! 
@strave und @omz Wenn Ihr Fragen habt, stehe ich gerne zur Verfügung.

Beim nächsten Treffen können wir gerne noch mal kurz die wichtigsten Punkte wiederholen, da wir sowieso noch ein paar Messungen machen müssen (z.B. das Verhalten mit eingeschaltetem Scheinwerfer).

jjac, 2019-03-25:
Ich (der designierte Protokollant) komme erst jetzt dazu, letzten Freitag zu verarbeiten. Eure sehr guten Darstellungen kann ich nur durch Erkenntnisse aus unserer kurzen Erörterung verschiedener Programmierumgebungen ergänzen.

@NikolaiRuhe hat sich bereits mit der Programmierumgebung zum Nordic Semi nRF51822 (?) beschäftigt. Erkenntnisse:
- über einen (zusätzlich zu erwerbenden) USB-Adapter am Mac lassen sich Programme ohne boot loader per Kabel (4 Anschlüsse) auf den Chip übertragen. Der USB-Adapter nutzt dabei eine Eigenschaft des Nordic Semi, die es erlaubt, mit hohen Spannungen (ca .11 V (?)) den Flash-Speicher zu „brennen“.
- Ebenfalls unter Verwendung des USB-Adapter und der zur Verfügung stehenden tool chain unterstützt der Nordic Semi single-step debugging per GDB direkt auf dem Chip

@ole zeigte kurz die Arduino-Standard-Entwicklungsumgebung. Erkenntnisse:
- Programmierung in C++, ohne Notwendigkeit der Inkludierung von header files
- als einzige Debugging-Option steht das sog. printf-Debugging zur Verfügung
- keine Unterstützung von code completion
- bei Problemen steht ein reichhaltiges Angebot von Lösungsvorschlägen im Internet zur Verfügung

Last but not least haben wir etwas über die Hilfsbreitschafts-Nettigkeit (HN) in verschiedenen Internet-Gemeinschaften gelernt. Es gilt:

HN_Elektroniker < HN_Programmierer < HN_Fotograf
(Satz des Ruhe)