savoyhardwaregroup/speedometer #20: Marktübersicht Oszilloskope

ole, 2019-03-25:
Ich habe mich ja ein bisschen informiert, welches günstige Oszilloskop man kaufen sollte. Ist vielleicht auch für andere interessant, deswegen schreibe ich das hier mal auf. Alles ohne Gewähr, ich habe keines der Geräte selber gesehen oder gar benutzt. Außerdem habe ich ja gar keine Erfahrung, kann mir also gar kein fundiertes Urteil erlauben. Ergänzungen sind willkommen!

Ich habe mir hauptsächlich Geräte mit 4 Kanälen angesehen. Die meisten Modelle gibt es auch mit 2 Kanälen, dann sind sie billiger.

Fast alle Hersteller benutzen konsistente vierstellige Modellnummern, was Vergleiche einfacher macht. Muster: SBBK, wobei:

* S = Serie (1 = Einsteiger, 2, ... , 7 = Profi)
* B = Bandbreite (2 Ziffern), z.B. 05 = 50 MHz, 10 = 100 MHz
* K = Anzahl Kanäle (2 oder 4)

Für viele Geräte gibt es Hacks, mit denen sich die Bandbreite erhöhen lässt und teilweise weitere Features freischalten lassen (z.B. Logic Analyzer oder ein Funktionsgenerator). Die Hersteller verkaufen dieselbe Hardware in sehr unterschiedlichen Preiskategorien. Die billigen Modelle sind per Software begrenzt. Das betrachte ich hier aber nicht.

### [Hantek MSO5074FG](https://www.amazon.com/Hantek-MSO5074FG-Oscilloscope-Channels-Generator/dp/B00HEHUDB8/)

[image]

Das hier ist @NikolaiRuhe's Gerät zum Vergleich.

* 4 Kanäle
* 70 MHz
* 1 Gigasample/s
* 1 Megapoints Speichertiefe
* Display: 7 Zoll, 800×480

"MSO" steht für "Mixed Signal Oscilloscope", d.h. es hat auch einen Logic Analyzer für 8 oder 16 digitale Kanäle. Und "FG" steht (glaube ich) für einen eingebauten "Function Generator". Beides sind i.d.R. aufpreispflichtige Extras, die es für jedes Modell gibt, die aber in den hier genannten Preisen nicht drin sind.

### [Rigol DS1054Z](https://www.rigol.eu/products/digital-oscilloscopes/1000z/)

[image]

* 380 € (alle Preise inkl. USt) ([Beispiel-Händler](https://www.batronix.com/versand/oszilloskope/Rigol-DS1054Z.html))
* 4 Kanäle
* 50 MHz
* 1 Gigasample/s
* 12 Megapoints Speichertiefe
* Display: 7 Zoll, 800×480

Das ist der Klassiker, den die meisten Leute empfehlen. Sieht nach einem soliden Einsteiger-Gerät aus. Im Gegensatz zu Nikolais Gerät fällt auf, dass es nur einen Vertikal-Regler für alle Kanäle gibt und nicht einen pro Kanal. Man muss also erst den richtigen Kanal aktivieren, bevor man den Regler benutzen kann. Das ist bei den meisten Geräten so.

Nachteile: Modelljahr 2014/15, also schon ziemlich alt. Vielleicht gibt es bald einen Nachfolger? Ich finde auch das Design nicht besonders hübsch, aber das ist Geschmacksache.

### [Siglent SDS1104X-E](https://www.siglentamerica.com/digital-oscilloscopes/sds1000x-e-series-super-phosphor-oscilloscopes/)

[image]

* 510 € ([Beispiel-Händler](https://www.batronix.com/versand/oszilloskope/Siglent-SDS1104X-E.html))
* 100 MHz
* 1 Gigasample/s
* 14 Megapoints Speichertiefe
* Display: 7 Zoll. 800×480

Sehr ähnliche Specs wie das Rigol, bis auf die höhere Bandbreite. Es macht in Reviews auf mich aber einen besseren Eindruck: bessere Software, schnelleres UI (vielleicht hat es einen schnelleren Prozessor), etwas schöneres Design (Nikolais Hantek finde ich aber schicker). Auf jeden Fall ist es ein deutlich neueres Modelljahr (2017).

Wenn ich das richtig sehe, kann das Siglent diverse serielle Protokolle (UART, I²C) out of the box decoden.

Vielleicht ein Vorteil: das Teil hat einen eingebauten Webserver, mit dem man es über einen Webbrowser steuern kann. Ich habe ein Video gesehen, in dem das überraschend flüssig ging. Braucht man das? Keine Ahnung. Für Screenshots etc. ist es vielleicht ganz nett.

Ich würde wahrscheinlich die 130 € mehr ggü. dem Rigol ausgeben und zum Siglent greifen. Ist aber nur so ein Gefühl.

---

### Außer Konkurrenz: [Rohde und Schwarz RTB2004](https://www.rohde-schwarz.com/de/produkt/rtb2000-produkt-startseite_63493-266306.html)

Hier ein Beispiel von einem viel teureren Markengerät, das mir aufgefallen ist, weil ich es unglaublich schick finde. Sicher kein Vergleich zu den anderen Modellen, aber auch eine völlig andere Preisklasse.

[image]

* 2260 € ([Beispiel-Händler](https://www.batronix.com/versand/oszilloskope/Rohde-Schwarz-RTB2004.html)). Die 2-Kanal-Version kostet immer noch 1500 €.
* 4 Kanäle
* 70 MHz
* 1,25 Gigasamples/s pro Kanal (bzw. 2,5 bei 1–2 aktiven Kanälen)
* 10 Megapoints Speichertiefe pro Kanal (bzw. 20 bei 1–2 aktiven Kanälen)
* **Display: 10 Zoll, 1280×800, Touchscreen**
* **Besonderheit: 10-Bit A/D-Wandler** (Standard ist 8 Bit)

Das Display ist viel größer und höher aufgelöst. Selbst der Touchscreen scheint gut zu funktionieren (Pinch to Zoom etc.), wenn auch sicher nicht ganz auf iPad-Niveau. Ein sehr gutes Web-Interface hat es auch. Am besten finde ich, dass man eine USB-Maus anschließen und die Menüs dann per Maus bedienen kann. Ich habe das in einem Video gesehen und es sah sehr schnell und komfortabel aus, besonders wenn das Gerät erhöht steht so wie bei Nikolai.

Nachteil: spiegelnder Bildschirm.

NikolaiRuhe, 2019-03-25:
Ich würde definitiv noch die neue [Rigol MSO5000](https://www.rigolna.com/products/digital-oscilloscopes/mso5000/) Serie betrachten, besonders das MSO5074. Das ist bei mir ganz oben auf der Kauf-Liste. Die MSO5000er benutzten den supermodernern "Phoenix" ASIC und sind wohl eine recht zukunftssichere Investition.

[image]

- 1070 € ([Beispiel-Händler](https://www.batronix.com/shop/oscilloscopes/Rigol-MSO5074.html)).
- 4 Kanäle
- 70 MHz
- 2 Gigasamples/s pro Kanal (bzw. 4 oder 8 bei 1–2 aktiven Kanälen)
- 25 Megapoints Speichertiefe pro Kanal (bzw. 100 bei 1 aktiver Kanal)
- Display: 9 Zoll, 24Bit 1024×600, capacitive Touchscreen

Hier ist die [Besprechung](https://www.eevblog.com/2018/11/12/eevblog-1146-new-rigol-mso5000-oscilloscope/) von Dave (EEVblog) L. Jones.

NikolaiRuhe, 2019-04-07:
Ich plane, demnächst auf obiges Rigol umzusteigen und würde mein Hantek für 300 Euro verkaufen. Ich hoffe, das wird in den nächsten Tagen/Wochen konkret. Falls also jemand Interesse hat, meldet Euch. (Jörg hat sich schon mal angemeldet)

ole, 2019-04-20:
### [GW Instek GDS-1054B](https://www.gwinstek.com/en-GB/products/detail/GDS-1000B)

[image]

* 550 € ([Beispiel-Händler](https://www.datatec.de/oszilloskope/oszilloskope-bis-500-mhz/gds-1054b-oszilloskop-4-kanal-50-mhz-1-gsa/s-50.000-wfm/s-10-mpts-usb-lan?c=30225693))
* 50 MHz
* echte 1 Gigasample/s pro Kanal (keine Aufteilung bei mehreren aktiven Kanälen)
* Speichertiefe: 10 Megapoints pro Kanal
* Signalerfassungsrate: 50.000 Waveforms/s pro Kanal
* Display: 7 Zoll. 800×480

Das hier kommt in Reviews auch ziemlich gut weg. [Teardown bei EEVBlog](https://www.youtube.com/watch?v=9SXZk2tsQ4g).

Pro:

* Separate Drehköpfe für jeden Kanal
* Etwas bessere technische Daten als die (etwas billigere) Konkurrenz
* Responsives UI

Contra:

* Sieht ein bisschen antiquiert aus. Der kleine Bildschirm verliert sich in dem vergleichsweise großen Gehäuse ein bisschen.
* Keine Zusatzfeatures wie Serial Decoding oder eingebauter Webserver.

Modelljahr: 2015/16 (glaube ich).