savoyhardwaregroup/speedometer #22: Toolchain Installation

NikolaiRuhe, 2019-03-29:
Verwendete Komponenten:

- ARM embed
- ARM gcc embedded toolchain (gcc, gdb, c und c++ libraries)
- OpenOCD
- make, homebrew

wip

NikolaiRuhe, 2019-03-29:
## 1. ARM mbed

ARM mbed funktioniert online im Browser oder lokal in der Shell. Wir richten zuerst den Online-Zugang ein, richten ein Projekt ein, exportieren das dann als lokales Projekt und benutzen später nur noch das Kommandozeilen-Interface.

Hier sind die benötigten Schritte der Reihe nach:

1. [ARM mbed Account](https://os.mbed.com) auf der Website anlegen.
2. Online Compiler öffnen
3. Nordic nrf51822 Target hinzufügen
4. Beispiel-Projekt importieren
5. Projekt exportieren
wip

jjac, 2019-04-01:
## 2. ARM GCC Embedded Toolchain

1. Download  `gcc-arm-none-eabi-8-2018-q4-major-mac.tar.bz2` from [(https://developer.arm.com/tools-and-software/open-source-software/gnu-toolchain/gnu-rm/downloads)]
2. follow instructions documented on [https://gnu-mcu-eclipse.github.io/toolchain/arm/install/]
3. $ brew install sRecord
4. To serve as an example import `Blinking-Led` into your web compiler IDE from [https://os.mbed.com/teams/Icarus-Sensors/code/Blinking-Led/]
5. Export the project to your Downloads folder (don’t miss to choose `Make-GCC-ARM` within the export dialogue)
6. Within the project’s root directory type `make`
7. If make succeeds with message „hex file ready to flash“ your build tool chain is all setup

jjac, 2019-04-01:
## 3. OpenOCD

1. `$ brew install openocd`
2. Connect development board via USB-Adaptor to Mac
3. Start OpenOCD server: `$ openocd  -f interface/stlink-v2.cfg -f target/nrf51.cfg`
4. Start console to OpenOCDControl server `$ nc localhost 4444`
5. `> halt`
6. If your nrf51 SoC's low-end flash memory sections are yet memory protected issue command `> nrf51 mass_erase`
6. `> flash write\_image erase <PathToCombinedHexfile>`
7. `> reset`
8. Voilà, your dev board LED 0 should be blinking

jjac, 2019-04-01:
## Schematic Drawing of Communication Setup

[image]

NikolaiRuhe, 2019-04-01:
OpenOCD ist gleichzeitig:

- der USB Hardware-Treiber des ST-Link v2
- und Debug Server für
    - GDB (auf Port 3333, aus gdb per `target remote localhost:3333` erreichbar)
    - Hardware Control und Flash (auf Port 4444, z.B. per `nc localhost 4444` erreichbar)

NikolaiRuhe, 2019-04-01:
> ## 2. ARM GCC Embedded Toolchain
> Download gcc-arm-none-eabi-8-2018-q4-major-mac.tar.bz2 [...]

Die ARM gcc toolchain kann man auch gut durch Homebrew installieren:

    $ brew install gcc-arm-none-eabi
    $ arm-none-eabi-gcc --version
    > arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 [...]

ole, 2019-04-05:
Um den Debugger zu starten:

```
$ arm-none-eabi-gdb
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git
...
(gdb) target remote localhost:3333
(gdb) file Build/Blinking-Led.elf
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
(gdb) list
(gdb) break main.cpp:9
Breakpoint 1 at 0x161b8: file ../main.cpp, line 10.
(gdb) continue
Continuing.
(gdb) next
(gdb) info break
Num     Type           Disp Enb Address    What
1       breakpoint     keep y   0x000161b8 in main() at ../main.cpp:10
	breakpoint already hit 1 time
(gdb) delete 1
(gdb) continue
Continuing.
```

Irgendwas ist noch falsch an den Zeilennummern. Die Debug-Informationen scheinen nicht 100% mit dem Sourcecode übereinzustimmen.

jjac, 2019-04-05:
## printf Debugging ermöglichen

1. [USB to UART Driver für den SILABS CP2102](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers) Chip auf unserem BLE400 Board herunterladen

NikolaiRuhe, 2019-04-07:
Der Silicon Labs CP2102 ist ein sehr verbreiteter USB to UART bridge controller. Er ist in vielen (günstigen) developer boards integriert; daher macht es durchaus Sinn, den Treiber zu installieren.

Es gibt noch zwei weitere verbreitete Chips:

- FT232 von Future Tech (schottischer Hersteller, teuer, sehr verlässlich, z.B. im original Arduino)
- CH340 von WCH (chinesisch, sehr günstig)

NikolaiRuhe, 2019-04-07:
Als Software auf dem Mac empfehle ich [Serial](https://www.decisivetactics.com/products/serial) (statt Terminal).

Das kann alle Emulationen, kennt die Chips und ihre Treiber, repariert Bugs bei der Flusskontrolle (DTS, CTS), kurz: es funktioniert einfach, wenn man im Terminal noch den ganzen Tag nach der richtigen Einstellung sucht.

ole, 2019-04-07:
> Als Software auf dem Mac empfehle ich Serial (statt Terminal).

Danke. Serial hatten wir auch gefunden. Das funktioniert sogar ohne den CP2102-Treiber.

ole, 2019-04-15:
Ich habe meinen Stand jetzt gepusht. Es gibt ein neues Verzeichnis namens [`code`](https://github.com/savoyhardwaregroup/speedometer/tree/master/code) mit einer Readme-Datei mit rudimentären Build-Instruktionen. Schaut mal, ob es bei euch funktioniert.

(Spoiler: das von CMake generierte Xcode-Projekt scheint uns nicht viel zu nützen, weil Xcode mit dem GCC-Compiler nicht klarkommt.)

NikolaiRuhe, 2019-04-18:
Auf sparkfun gibt es ein [Tutorial](https://learn.sparkfun.com/tutorials/efficient-arduino-programming-with-arduino-cli-and-visual-studio-code) zu VS Code und Arduino. Das ist zwar für Windows und Arduino – ich habe aber trotzdem was gelernt.

nsc, 2019-04-18:
@ole Sieht gut aus. Bei mir funktioniert es. Ich musste noch `brew link cmake` machen, weil ich schon ein cmake rumliegen hatte, aber dann ging es.