savoyhardwaregroup/speedometer #23: 74xx Logik-Chips: LS vs. HC vs. HCT

ole, 2019-04-07:
Ich will mir demnächst ein Sortiment an [Logik-Chips](https://en.wikipedia.org/wiki/7400-series_integrated_circuits) kaufen, also AND- und OR-Gates und so ([Beispiel bei Segor](https://www.segor.de/#Q=74HC02N&M=1)). Ich bin mir nicht ganz sicher, [welche Chip-Familie](https://en.wikipedia.org/wiki/7400-series_integrated_circuits#Families) aus der 74xx-er-Reihe am besten ist.

Ben Eater benutzt in seinen Videos die LS-Reihe (TTL), es gibt aber auch CMOS-Baureihen: HC und HCT, die ja wohl insgesamt moderner sind und mit kleineren Strömen arbeiten.

Ich hatte zur HCT-Reihe tendiert, weil die kompatibel zu den alten TTL-Teilen sind, aber ein anderes Einsteigerbuch (Make: Electronics) empfiehlt HC.

@NikolaiRuhe Kannst du etwas empfehlen? Es geht mir um einfache Breadboard-Experimente und evtl. nach und nach sowas wie den 8-Bit-Computer von Ben Eater nachzubauen.

ole, 2019-04-07:
Der Vollständigkeit halber: ich habe auch gelernt, dass es noch die [4000-er Serie](https://en.m.wikipedia.org/wiki/4000-series_integrated_circuits) gibt. Ich habe erstmal an die 7400-er gedacht, weil Ben Eater die verwendet, aber vielleicht sind die 4000-er die bessere Wahl?

NikolaiRuhe, 2019-04-07:
Die 4000er haben den Nachteil, dass sie sehr leicht durch ESD Schaden nehmen. Das ist ein klarer Nachteil für Steckbrett und Prototyping, da man sie da oft in die Hand nimmt. Dafür haben sie einen sehr großen Bereich an erlaubter Betriebsspannung (bis 20V).

Ich habe immer mit den HCT Typen gearbeitet. Allerdings ist das lange her, da es heute selten Sinn macht, diskrete Logikbausteine zu verwenden. Es gibt natürlich Ausnahmen und für Lernzwecke sind sie auch super. Aber in Produkten trifft man sie kaum noch an. MCUs und FPGAs können einfach so viel mehr.

Du solltest auch berücksichtigen, mit was Du die Bausteine verbinden willst. TTL Level sind heute bei MCUs nicht mehr sehr verbreitet, also machen vielleicht die AC/HC Typen mehr Sinn (2-6V Betriebsspannung).

Die AC Typen können mehr Last treiben, wenn Du mehr als eine LED an die Ausgänge anschließt (“fanout”).

Hier ist noch ein [Vergleich](http://solarbotics.net/bftgu/starting_elect_ic.html).

ole, 2019-04-08:
Danke, das hilft mir schon weiter!

closed 2019-04-09 by ole