savoyhardwaregroup/speedometer #25: Laborausstattung

NikolaiRuhe, 2019-04-12:
Ungeordnete Liste praktischer Einkauftipps:

- Header: einreihig (gerade, verschiedene Farben, abgewinkelt) [Aliexpress](https://m.aliexpress.com/item/32873263301.html), [Aliexpress](https://m.aliexpress.com/item/32882291626.html)
- Lochrasterplatinen: FR4 (aus Epoxy, nicht Hartpapier), durchkontaktiert, eher kleine Maße. [Aliexpress](https://m.aliexpress.com/item/32888868148.html)
- Widerstände: Sortiment ca. 500 Stück, 1%, 1/4 W [Aliexpress](https://m.aliexpress.com/item/32840627690.html)
- Elektrolytkondensatoren: Sortiment ca. 100 Stück [Aliexpress](https://m.aliexpress.com/item/32866006892.html)
- Scheibenkondensatoren: Sortiment ca. 200 Stück [Aliexpress](https://m.aliexpress.com/item/32864984481.html)

WIP, Liste wird fortgesetzt: Leuchtdioden, Transistoren, OpAmps, Steckverbinder, Schalter, Schrumpfschlauch, Lötkolben, Entlötpumpe, Lötzinn, Kabel, ...

ole, 2019-04-12:
Super, danke fürs Anfangen. Ich habe den Seitenschneider und meine Abisolierzange ergänzt.

ole, 2019-05-12:
Ich habe die Links zu den Sortimentskästen mal hier ergänzt.