savoyhardwaregroup/speedometer #27: Projektidee: Funken mit LoRaWAN

ole, 2019-04-20:
[LoRaWAN](https://de.wikipedia.org/wiki/Long_Range_Wide_Area_Network) ist ein relativ neuer Mobilfunkstandard für stromsparende Kommunikation mit sehr niedrigen Datenmengen:

> Die Reichweiten erstrecken sich von 2 km (Stadtgebiet) über 15 km (Vororte) bis zu 40 km (ländliche Gebiete). Ein weiterer großer Vorteil ist die Durchdringung von Gebäuden, da hier auch zu einem gewissen Grad unterirdische Räumlichkeiten versorgt werden können. Der Strom von Endgeräten beträgt rund 10 mA und 100 nA im Ruhemodus. Das ermöglicht je nach Anwendungsfall und unter Vernachlässigung der Selbstentladung, eine Batterielebensdauer von 2 bis 15 Jahren. Die Kommunikation zwischen den Endgeräten und den Gateways erfolgt auf verschiedenen Frequenzkanälen mit unterschiedlichen Datenraten. Diese liegen zwischen 0,3 kbit/s und 50 kbit/s.

In Berlin scheint es bereits ein Community-betriebenes ([The Things Network](https://de.wikipedia.org/wiki/The_Things_Network)) und kostenlos nutzbares LoRaWAN-Netz zu geben, an dessen Ausbau man sich natürlich auch selbst beteiligen kann, indem man sich selbst ein Gateway aufs Fensterbrett stellt.

Wenn ich das richtig verstehe, bauen aber auch Telekom und Co. solche Netze auf, in die man sich dann für sehr geringe Gebühren einmieten kann.

Einen konkreten Anwendungsfall habe ich nicht, aber die Technologie interessiert mich. Klassische Anwendungen sind sicher Umweltsensoren (Wetter, Feinstaub, Stickoxide). Im Zusammenhang mit dem Thema Fahrrad wäre es vielleicht für eine Verfolgung im Fall eines Diebstahls interessant?

ole, 2019-04-20:
Hier noch ein Talk zum Thema vom Chaos Communication Congress 2018: https://media.ccc.de/v/35c3chaoswest-27-long-range-radio-communication-for-everyone-using-lorawan- (ich habe das Video noch nicht angeschaut).

NikolaiRuhe, 2019-04-24:
LoRaWAN interessiert mich auch sehr. Als Testprojekt hatte ich die Idee, ein GPS-Modul mit LoRa auszustatten. Damit hätte man ein Gerät, das (zumindest draußen) immer wieder auffindbar ist. Anwendungsbeispiele wären z.B. Fahrraddiebstahlschutz oder Auto-park-Finder.